KimJorgensen/KungFuFlash #183: Insert SD-card issue

sonkite, 2023-12-22:
I have recently ordered 10 KFF v2 from jlcpcb.
Mounted all the components and flashed successfully with 1.48.

It works fine - for a while.... for all of them!

After that, like 30min plus, it says "insert fat formatted sd card".
Same issue for so far all of them except one, i think... havent fully tested all.

If i switch sdcard it doesnt help. 

After issue presented it comes back earlier next time...
Exchanged sdcard reader on one.. 

Any suggestions? All 10 cant be faulty i presume...

Tested on 2 different c64c.
Testef 2 different sdcards.

Thanks in advance 
Johan

KimJorgensen, 2023-12-23:
Hi Johan

Let me see if I understand what is happening: After using the Kung Fu Flash cartridge for 30+ minutes then it won't read the SD card anymore.
But then what? You reset the cartridge and it works again? Or you power off the C64 and it works again? Or the sdcard is totally dead and won't work again, but a new sdcard will until that also dies after 30+ minutes?

sonkite, 2023-12-23:
Sorry for being unclear.

It works for 30min.
After that it doesnt detect the card anymore.
Doesnt matter if I reset etc....only thung that helps it cooling off.
Sdcard works fine if I put it in the computer or another kff cart.

Same issue with all my 10 build kff carts.

Seems like some component is weakening with heat / current and I dont know where to look. Seems wierd that all 10 has the same issue.

Tested 2 different sdcards, 10 kff selfbuild (from same batch).

First test of a kff cart, it works for lika 30-60min.
After that it works less for every time, after some hours of cooldown

Tested 2 different c64c. Same problem so I dont think its the computer.

KFF diagnostics says Clock frequency at 985220.

Dont know where to look for failures. Some soldering? Some weak components?

KimJorgensen, 2023-12-24:
I have never experienced anything like that, and none of the components on the Kung Fu Flash cartridge seems to get hot at all.
Have you tried with a different power supply?

sonkite, 2023-12-26:
I have 1 original PSU (Light Grey version) and 1 brand new modern switched PSU. Tested both, same issue.

I have now made a new rest.
2/8 KFF works perfectly for 2h+.
6/8 KFF fails within 1,5h. After cooldown (waited like 3h, didnt check earlier) works again but eventuelly fails again within the same timeframe.

Tested them with same 2 sdcards, both computers, both PSU - no pattern as i can see.

I have replaced sdcard reader on 1 but sadly left it at work for the holidays so cant test it...

Sdcard reader bought at Aliexpress, weak quality?

Any soldering i should check except for the obvious ones at the cardreader? :)

Merry christmas,
Johan

pmandes, 2023-12-26:
Check the voltage drop after heating on the LDO, whole faulty batches happen. Show a photo of your boards. Reflow MCU will not hurt if it is a cold solder issue.

Best
Pawel

sonkite, 2023-12-26:
Thanks for your answers, both of you :)

Here comes the pictures.
I will try to measure tomorrow and reflow mcu.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

sonkite, 2023-12-26:
Now measured U2.
From C64        After regulator U2
5.03V              3.31V       -- at start up
5.05V              3.31V       -- when failing the sdcard

"Right side of C2" - 3,31V
"Right side of C4" - 3,31V

UffeJakobsen, 2023-12-28:
Are the SDcards getting hot ?
Your test description does not tell if you also let the SDcard rest for hours before testing another unit ?
I come across some (fake?) sdcards that got extremely hot and eventually stopped working... the would seem work again after cooling down...

sonkite, 2023-12-30:
Its the same two SD-cards im using.
Dont let them cooldown and problem is not that since it works in another KF.

It might be the SD-card readers, have noticed that strange behaviors... tried to resolder and works temporarily sometimes. Tried to replace and got better before worse again (readers from the same batch).

I have ordered new readers but takes like a month...

CqaZ80, 2024-01-04:
rinse with white benzine for 1 hour ( into a glass container) and clean with a toothbrush from time to time... I got some strange behavior pretty similar to you, some flux remains can do some problems if you don't clean enough.

sonkite, 2024-01-04:
Thanks for your reply.

I've washed in ipa (isopropyl alcohol) and tried to reheat the connections... i will try another bath for one of the cards :)

sonkite, 2024-01-19:
Update:
It got better with 4h ipa bath and cleaning with a toothbrush every 30min :)
It was probably not enough with my first cleaning which i thought was very carefully.

closed 2024-12-29 by KimJorgensen