KimJorgensen/KungFuFlash #187: ef3usb.exe not working on 64bit system

stefagnani, 2024-02-07:
I've tried to execute ef3usb.exe modified for KungFuFlash on Windows 10 64bit, the program fail to start. There's a workaround for this problem?

KimJorgensen, 2024-02-08:
There shouldn't be a problem running the program on Windows 10 64bit.
How did you try to execute the program, with which arguments, and what is the error message?

mvancopp1, 2024-03-02:
I have been having the same issue but it is v1.49 that breaks it. 

When using usbef3.exe comXX s the transfer will start, the c64 screen changes (looks like Basic color with no characters, blank screen), and the transfer never completes or errors out. It just hangs. If I downgrade to 1.48 it works like a charm.

Tried:
2 PCs: One Win10 64bit and one Win11 Pro 64bit
2 KFFs
Multiple usb cables
Redownloaded from GIT 1.49 just in case...
1 C128 PAL machine
1 C64 NTSC

1.48 always works and 1.49 always fails. 100% reproducible.

Let me know if you need different info or a specific test. I am happy to help

Thanks,
Mark

KimJorgensen, 2024-03-09:
Thank you for reporting.
It was indeed a bug in the v1.49 release, that has been fixed in v1.50

closed 2024-03-09 by KimJorgensen