KimJorgensen/KungFuFlash #188: Write my basic to SD card

8Potoooooooo, 2024-02-20:
Am I able to write my programs to the SD card in the King Fu Flash Cartridge?
If so, how does this work? I wasn't able to find the answer here.

marko-radojcic, 2024-02-21:
Same here, I can sometimes copy files from SD2IEC (at device 9) to Kung Fu Flash (at device 8 - but only if it's 8) and it doesn't work every time, sometimes it says "device not present error"... ?

KimJorgensen, 2024-03-01:
The Kung Fu Flash drive emulation must be active before you can to save a BASIC program to the SD card.

To activate the the drive emulation you can either mount a directory or disk image from the file options menu (SHIFT + RETURN), or you can launch a PRG file directly from the SD card, or from within a disk image.

The drive emulation will **not** be active if you exit to BASIC with the F keys, or while emulating cartridges

marko-radojcic, 2024-03-16:
Thank you @KimJorgensen. I tried copying from SD2IEC using DC64, and it properly sees the contents of Kung Fu Flash SD card. It proceeds with the copy operation but the file is not there. I will however try mounting a directory first. I tried this on root of the cartridge's SD card.

KimJorgensen, 2024-03-16:
@marko-radojcic I was talking about saving programs directly from BASIC, not from a copy program.

I suspect the issue you are having is that DC64 is not using kernal vectors for the copy operation. Instead it is probably using direct hardware access to the serial port, which the Kung Fu Flash in not connected to.

It would be nice if I could recommend a good copy program that both support Kung Fu Flash and regular Commodore drives with disk turbo. However, I have not being able to find one.

closed 2024-12-30 by KimJorgensen