KimJorgensen/KungFuFlash #192: Kung Fu Flash doesn't work properly on the KU-14194HB board

mvbhz, 2024-04-28:
I have a Commodore 64 from 1982 and a KU-14194HB board. Unfortunately, when starting Kung Fu Flash, an image of the file browser appears for a moment, but after a while it disappears and the BASIC screen appears without any characters. I tried with different firmware versions but it still doesn't work. Please improve the compatibility of KFF with the KU-14194HB board.

KimJorgensen, 2024-04-28:
Sorry, but I don't have such a board for testing, and I don't know if this is a general problem with that board revision.

Did you try to adjust the [phi2 clock offset](https://github.com/KimJorgensen/KungFuFlash/blob/master/firmware/README.md#diagnostic)?

mvbhz, 2024-04-28:
phi2clock cannot be set because it crashes to the BASIC screen without characters.

KimJorgensen, 2024-04-28:
Did you test your Kung Fu Flash cartridge with other Commodore 64 board revisions?

I'm wondering if KU-14194HB has the same [unusual reset circuit](https://www.lemon64.com/forum/viewtopic.php?t=46277#p556139) as the 326298 rev A. If the Kung Fu Flash cannot reset the C64 that could explain your issue.

mvbhz, 2024-04-28:
It works on a next motherboard from 1983.

So there is no solution for KU?

modeler, 2024-05-01:
I have a KU-14194HB here and Kung Fu Flash works just fine with it. You might have another issue with your board.

[image]

mvbhz, 2024-05-01:
Which revision, A or B? I have a B. But the board is functional, all tests are OK, programs and games work normally. If it's the motherboard's fault, I have no idea how to check what's damaged.

modeler, 2024-05-01:
> Which revision, A or B? I have a B. But the board is functional, all tests are OK, programs and games work normally. If it's the motherboard's fault, I have no idea how to check what's damaged.

Mine is a Rev. A, so it's still possible that it doesn't work with your particular board revision. These are relatively rare though, I've only ever seen this one.

mvbhz, 2024-05-01:
It started working. It may be a problem with a missing contact somewhere on the motherboard. At least I know Kung Fu Flash is compatible with this motherboard.

KimJorgensen, 2024-05-01:
A possible explanation is a bad connection to the cartridge, e.g. if the expansion port or contacts on the cartridge are dirty

mvbhz, 2024-05-01:
I had previously cleaned these contacts with IZO, and it still didn't work... I don't know what was the reason.

closed 2024-07-26 by KimJorgensen