KimJorgensen/KungFuFlash #194: Mach 5 cartridge support

Kopromaster, 2024-06-25:
Hello Kim, could you please help me with adding support of Mach 5 cartridge?

I've prepared changes in a fork

https://github.com/Kopromaster/KungFuFlash

nevertheless, I don't have development environment to build the firmware :-) Could you please shortly review my changes (cartridge is really simple) and build a binary for me? Then I'll test it in my C64 and if everything works, then I'll raise a pull request.

Thank you.

KimJorgensen, 2024-06-30:
MACH 5 support has now been added in the [latest release](https://github.com/KimJorgensen/KungFuFlash/releases/tag/v1.51) .
I did change the implementation to be based on the Warp Speed cartridge to make the firmware slightly smaller.

Thank you for your support

closed 2024-06-30 by KimJorgensen

Kopromaster, 2024-07-01:
Thank you for new release, it was amazing to see my cartridge working again after 30 years on real hardware and not just in emulator :-)