KimJorgensen/KungFuFlash #195: Mega65 C64 Core, odd p

Kugelblitz360, 2024-07-12:


KimJorgensen, 2024-07-13:
The Phi2 clock is generated by the C64 hardware and is not affected by the Phi2 offset in any way.

The KFF uses the Phi2 clock to know when to read the address bus, the control signals etc. Adjusting the offset will change when to read these signals.

Note that the KFF is not a precision instrument and variations between two different cartridges can be expected, see https://github.com/KimJorgensen/KungFuFlash/issues/89#issuecomment-923126640

So if the Phi2 clock is off you need to fix that in the Mega65 C64 Core

Kugelblitz360, 2024-07-13:
Yeah - issue is under discussion over there. Question is if you have a suggestion on setting the offset with the values I posted.

Thanks, Boris!

KimJorgensen, 2024-07-14:
There is no relationship between the phi2 clock frequency and the phi2 offset a C64 may require. It is dependent on the timing requirements of the hardware in the specific C64.

Sorry but I don't own a Mega65 so I can't offer you much advise. You can try increasing or decreasing the offset by one at a time, but I don't know if adjusting the offset would even solve your problem

Kugelblitz360, 2024-07-14:
The root cause was something totally different: The combination of JaffyDOS, KFF and the Mega65 does not work properly. Switching to original JiffyDOS all problems go away.

closed 2024-07-14 by Kugelblitz360