KimJorgensen/KungFuFlash #198: Black screen loading any multiload EasyFlash games from $olo1870 - C64C shortboard

mamejay, 2024-08-19:
I have a strange issue on my C64C.  Everything loads perfectly except for any games from $olo1870.
In the OneLoad64 collection, the majority of the EasyFlash mutliload games are from a dev call $olo1870 and I just get a black screen.
I have tried playing with the phi2 offsets but cannot get any of these to start.

Anyone experienced this?

KimJorgensen, 2024-09-08:
I tried some $olo1870 MultiLoad64 games from OneLoad64 Games Collection v4 and OneLoad64 Games Collection v5 on my C64C and didn't have any problems.

If you have access to another C64 you could test with that to see if it hardware related.

KimJorgensen, 2025-01-04:
Is this the same issue as #206?

mamejay, 2025-01-04:
Hiu Kim.  This can now be closed as its related to JiffyDos and running $olo1870's games.  If I disable it everything is fine.  $olo confirmed this is the case.

KimJorgensen, 2025-01-04:
OK, thank you for reporting back

closed 2025-01-04 by KimJorgensen