KimJorgensen/KungFuFlash #205: Measured phi2 wrong value

Morph26, 2024-11-14:
Hello.

I have C64C Assy 250469 Schematic 252311 Rev B PAL. Kung fu flash won't start - even menu(or flash not inserted message) not showing. Kung fu flash rev.2
Checked 8 pin of IC 8701 - 17.7344 MHz, phi2 on kung fu flash pcb - 985.216 kHz. Kung fu flash Y1 - 8.0001 MHz
Plugged usb to PC and connected by terminal to COM port. In diag mode measured freq - 984.946 kHz, maximum offset +20 (not enough).

KimJorgensen, 2024-11-14:
Does that Kung Fu Flash cartridge work with another C64? If you do that test the phi2 offset should be set to 0, 20 is way too much.

Note that the KFF is not a precision instrument and you cannot expect the phi2 measurement to be the same as e.g. your oscilloscope reports, see https://github.com/KimJorgensen/KungFuFlash/issues/89#issuecomment-923126640

Morph26, 2024-12-17:
Problem was in STM32F405. Replaced, reflashed - all good. Thank's.

KimJorgensen, 2024-12-17:
Great. Good that you got it working

closed 2024-12-17 by KimJorgensen