KimJorgensen/KungFuFlash #207: Add Ross format support

nikvla, 2024-12-08:
Ross format is used in many eastern European countries, and it would be nice to have support here.

nikvla, 2024-12-21:
There is a pull request to address this

TomCat666, 2025-02-11:
Thank you nikvla, hope it can be merged soon.

nikvla, 2025-02-18:
Not tried it on v2 as I have only v1 and it works there. I would suspect it might be an issue with initialization. On v1 it didn't work until I have removed cart init routine completely, it might be that on v2 it needs to be properly initialized.

TomCat666, 2025-02-18:
Thank you for coming back to me. I have uncommented the C64_CRT_CONTROL(STATUS_LED_ON|CRT_PORT_16K);  line in the init routine and it did not make any difference unfortunetly. Still black screen.

btw. there is a very cheap option to get KFF2 for under 10 EUR over on Forum64, you can apply in this thread:

https://www.forum64.de/index.php?thread/153660-kung-fu-flash-2-bausatz-sammelbestellung-runde-2/&postID=2254038#post2254038

It is well worth it.

KimJorgensen, 2025-03-23:
The  pull request has been merged and support for 64 kb ROM has been added

closed 2025-03-23 by KimJorgensen