KimJorgensen/KungFuFlash #208: Second 64KB memory space not cleaned when change .CRT file

lroby74, 2024-12-08:
Hello,
if i load a 128KB .CRT file, with intro on bank 8 (begin of 2' 64KB bank), when then i load another 64KB .CRT file, i see again intro of previous .CRT file.
I think that when i change .CRT file, second 64KB memory bank don't be cleaned.
I have some test files if needed
On Vice 3.8 this issue don't exists

KimJorgensen, 2024-12-11:
Sounds like a bug to me.
It would help if you could attach the test files or provide a link to these. Thanks

0wwafa, 2024-12-12:
Hello, the reason of the bug is quite simple: in a magicdesk/ocean/unicart (and alla variations) the "special" address DE00 manages the bank switching and enable/disable of the device BUT not as people think:

1) any address in the range DE00-DEFF is the same as DE00  so storing 1 in DE9A (for example) is the same as storing it in DE00.

2) the real bank read is BANK  && (MAX EEPROM BANKS-1)   so in a 64K eeprom, bank 8 = bank 0, bank 9 equals bank 1. in a 128K CRT instead, bank 32 = bank 0.. and so on.

That happens because only the lower 3 (or 4  or 5...) bits are considered and connected.

You can check that in the schematics of any cartridge.
[image]
[image]
[image]