KimJorgensen/KungFuFlash #210: Communicating with cart error

leepedley, 2024-12-28:
Hi,

new to all this and just purchased this cert but not getting very far.

on my c64 I am getting an error about it not communicating with the cart.

Tried holding the menu button down to get into the diagnostic menu but nothing appears to work.  The reset seems to work but nothing else.

my board is 2523|| REV.B if that’s any help.

Thanks

KimJorgensen, 2024-12-28:
It sounds like a hardware issue and I think you should contact the seller for support

leepedley, 2024-12-28:
Thanks Kim so it won’t be a timing issue?ThanksLeeSent from my iPhone

KimJorgensen, 2024-12-28:
I have no problem with people selling Kung Fu Flash cartridges, but they should also be willing to handle the support if something doesn't work.
If it turns out to be a general issue that hasn't been addressed, then let me know.

closed 2024-12-28 by leepedley