KimJorgensen/KungFuFlash #213: CRT files do not load

kmiii, 2025-01-26:
Hello,

My Commodore 64 does not load most of the .crt files, I would say that about 90% of them do not work, but the rest of the formats work correctly, I am currently loading the games in PRG format.

My Commodore's motherboard is 250425, I have tried to adjust the phi2 offset without success, is it possible that my motherboard has different timings?

KimJorgensen, 2025-02-18:
It's really hard to troubleshoot when an issue is described as generally as this. Which CRT files exactly and what do you mean by "do not work"?

If this is a bug report or feature request, please use the correct template when creating the issue

kmiii, 2025-02-18:
Hi,

I'm sorry I didn't express myself well, the .crt game files are the problem, when I say they don't work I mean I get a black screen in 90% of the .crt games.

It may just be a problem with my 250425 motherboard, that's why I was asking if there was any possibility of getting KFF to work with crt on my motherboard.

kmiii, 2025-02-20:
Yes, my Kung Fu Flash was tested with 2 Commodore 64s from friends, with different motherboards than mine, and it worked correctly with all the .crt games.

I also validated the .crts by transferring them from my SD to the PC and they worked fine with Vice.

On my Commodore 64 the only .crt games that work correctly are those in OneLoad64 v5 with the name [EasyFlash].crt, it's very strange.

KimJorgensen, 2025-02-21:
Has any modifications been made on your motherboard, like a custom kernal or replacement PLA?

Have you tested your C64 with other cartridges?

kmiii, 2025-02-21:
Yes, in the past the MOS8701 broke down and was replaced with a TOLB 1.2, all other components are original.

I was not able to test the C64 with another KFF, I only have one unit, but I do have a Dead Test cartridge that works fine.

KimJorgensen, 2025-02-22:
I don't know if the MOS8701 replacement can cause problems with the Kung Fu Flash. If possible, you should test with an original part to rule that out.

It could also be a problem with the expansion port and I would try to test with other cartridges

closed 2025-03-08 by KimJorgensen