KimJorgensen/KungFuFlash #223: USB data tranfer does not work

Steph71, 2025-08-09:
Thanks a lot for this nice cartridge, really great work.

**Describe the bug**
Trying to copy data from PC to KFF does not work. Basically already the USB data test fails.

**To Reproduce**
Steps to reproduce the behavior:
1. Switch on c64
2. press KFF menu Button (I hear UBS enumeration sound and COM5 shows up on device manager
3. on PC: .\ef3usb.exe COM5 0
4. On c64, I see "Communicating with USB"
5. Test never finishes

**Expected behavior**
Connection test passing

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]
[image]

**Kung Fu Flash (please complete the following information):**
 - Hardware: Rev2
 - Firmware: 1.5.3

**Computer (please complete the following information):**
 - Device:C64, C64C
 - Kernal: stock

**Additional context**
This used to work before ... probably something very obvious
Apart from not communication on usb ... everything works nicely

KimJorgensen, 2025-08-09:
The EasyFlash 3 USB test command (0) is not supported by Kung Fu Flash and will not work.
Did you try the send file command (s)?

Steph71, 2025-08-09:
stupid me :-)

forgot, that I had to send the server prg frist ... after doing that also usb test command works:

~~~
C:\Users\stephan\Downloads\KungFuFlash-master\KungFuFlash-master\3rd_party\ef3utils>.\ef3usb.exe COM5 s
EF3 USB Client v1.93.1 - Kung Fu Flash version

 - SEND PRG TO EF3 MENU

Handshake: EFSTART:PRG
Response: [LOAD]

 - Sending PRG file. Bytes sent:  13080

DONE!

C:\Users\stephan\Downloads\KungFuFlash-master\KungFuFlash-master\3rd_party\ef3utils>.\ef3usb.exe COM5 0
EF3 USB Client v1.93.1 - Kung Fu Flash version

 - TEST USB CONNECTION

Started Testing... (make sure all 4 jumpers are in DATA position!)
Testing was Succesfull ! Restart C64...
~~~

Thanks again for building this great cartridge.